MHeironimus/ArduinoJoystickLibrary #256: Not working in games, why?

TERM4X, 2022-11-24:
I've created a simple program (I'll attach It if necessar) on an Arduino Micro that sends values from 0 to 1023 about the X and Y axes at my computer. It works perfectly on the device manager and with online controller testers, but It dosen't work in any games (I've already test It on 4 differenti games controller-compatible)

Some suggestions?

MHeironimus, 2022-11-24:
Could you list some of the games that you have tested it on?

slawosz, 2023-01-15:
@TERM4X I had similar issues. After some digging, I managed to find that having too many options or too litle caused troubles in `Condor 2`. I really needed 1 axis and 1 button, and this is code that worked. Basically simple joystick. The game could not recognise device that had only throttle (only axis I am using ATM) - even though windows had no issues, it needed to look like a joystick.
```
#include <Joystick.h>

Joystick_ Joystick(
  0x03, // HID report ID
  JOYSTICK_TYPE_MULTI_AXIS,
  3, // Number of buttons
  0, // Number of hat switches
  true, // X axis
  true, //Y axis
  true, // Z axis
  false, // X rotation
  false, // Y rotation
  false, // Z rotation
  false, // Rudder
  true, // Throttle
  false, // Accelerator pedal
  false, // Brake pedal
  false // Steering wheel
);

int zAxis_ = 0;

const bool initAutoSendState = true;

void setup() {
  pinMode(2, INPUT_PULLUP);
  Joystick.begin();

}

int lastButtonState = 0;

void loop() {

  int currentButtonState = digitalRead(2);

  if (currentButtonState != lastButtonState) {
    Joystick.setButton(1, currentButtonState);
    lastButtonState = currentButtonState;
  }

  zAxis_ = analogRead(A0);
  zAxis_ = map(zAxis_, 0, 1023, 0, 255);
  Joystick.setThrottle(zAxis_);

  delay(50);
}
```