MHeironimus/ArduinoJoystickLibrary #258: Multi Joystick - USB device name issue with Unity Engine

Velenon, 2022-12-05:
<h1>Description of Issue</h1>
The Unity engine for Kerbal Space Program experiences a limitation whereby it identifies input devices by their reported name rather than a unique hardwareID therefore when trying to use 2No input devices with the same name the game will not recognise any inputs from one of the devices.
<h1>Current project</h1>
<p>My current project is to build a Multi Joystick control box for Kerbal Space Program(<b>KSP</b>), I could not get <b>KSP</b> to recognise all 6 axes required + the <b>"throttle"</b> axis as <b>KSP</b> would not recognise the libraries <b>"Throttle"</b> axis so I chose to create 2No Joystick devices using the example code given here and use the <b>"Rz Axis"</b> on the first joystick as the throttle axis.</br>

This all works fine and Windows sees and reports these axes doing what I expected, however both devices have the same name of <b>"Arduino Micro"</b> and for that reason only one of the Joysticks inputs are recognised by the game engine and passed to <b>KSP</b></br>

This project is being run on windows 10 for reference.</p>
<h1>Question</h1>
My question is this and please pardon my lack of knowledge when it comes to windows drivers and exactly what and how this library does what it does.</br></br>
But when creating/enumerating the two joystick devices or passing their information on as I assume this library does when connected, is it possible for two joystick names or numbers to be passed along to windows so that they can be indentified separately instead of having the same <b>Adruino Micro</b> name, or is the name shown in the windows USB game controller screen controlled entirely by the USB chip on the Arduino Micro?</br></br>
Thank you for any help you can give.

darkmos, 2022-12-29:
You need to modify the boards.txt file and change the line 
`micro.build.usb_product="Arduino Micro"` to `micro.build.usb_product="Arduino Micro Joystick 01"` for example. After you're done uploading the sketch to the first board modify again the line to `micro.build.usb_product="Arduino Micro Joystick 02"` and upload the sketch to second board.
Note there is a cleaner way to edit the boards.txt to add sub-menus variants to a particular board if you're often tinkering with your devices, there are some examples quickly found on Goggle.

Velenon, 2023-01-09:
Thanks Darkmos, this is indeed how I ended up doing this, though I had to use 2 separate arduino boards to be able to do it, I was hoping there was a way to achieve it when running the multiple joysticks instances from the single arduino board but it seems not.

marcelkraus, 2023-03-12:
I am sorry to chime in after two months, but does one have an idea how to find this file on a Mac? I would be very happy to modify the device name for my button box… ;) Thanks a lot!