MHeironimus/ArduinoJoystickLibrary #269: Lags and missing events on Raspberry Pi Zero W

Pilou44, 2023-02-04:
## Description of Issue
I've created a arcade stick using this library version 2.1.1 and an Arduino Micro. It works great on my Windows 11 PC and on a Raspberry Pi 3. But if I connect it to my Raspberry Pi Zero, I have input lags and missing events if button are not pressed long enough. I can see it just by running `cat /dev/input/event0`

## Technical Details

- Arduino Board (e.g. Arduino Leonardo):  Arduino Micro
- Host OS (e.g. Windows 10): Raspberry Pi Zero running Retropie or Raspbian
- Arduino IDE Version (e.g. 1.8.3): 1.8.19

Code available here : https://github.com/Pilou44/ArduinoJoystick/blob/main/stick_arcade_1_player/stick_arcade_1_player.ino

## Wiring Details
[image]

## Additional context
Add any other context about the problem here.

MHeironimus, 2023-02-04:
In looking at `stick_arcade_1_player.ino`, my guess is the **Arduino** is sending too much data to the **Raspberry Pi Zero**. I would recommend increasing the delay to 50 milliseconds (i.e., `delay(50);` and only calling `pressButton` or `releaseButton` when the value actually changes. You will want to keep track of the last know state of each button or axis and only call `pressButton` or `releaseButton` when the state changes.

Pilou44, 2023-02-05:
That's it! Thanks a lot. Now I save button states and use setButton only when necesary and it works great. I've increased delay to 30ms, I was afraid of missing frames with 50ms (even if I'm not sure difference would be humanly  perceptible).
Thank you very much.

closed 2023-02-05 by Pilou44