MHeironimus/ArduinoJoystickLibrary #271: Communication problem

dimigav, 2023-02-13:
Dear all,

I'm not sure if this is an issue. Please correct me if I'm wrong.
I made an flight controller with 2 axis (implemented by 4 loadcells and their amplifiers) and 23 buttons (using 3 SN74HC165 PISO configuration as @MHeironimus suggests). My problem is that everything works fine until I decide to close the Arduino IDE software. Then the controller identified by Windows as Game controller but no action when I press a button or moving the stick received.
I checked the initAutoSendState variable that is TRUE. What else?

MHeironimus, 2023-02-13:
It is hard to say without seeing your code, but one possibility is you are using the serial port in such a way that it requires the Serial Monitor to be running. If you have any code that is writing out to the serial port, try commenting it out.

dimigav, 2023-02-13:
> It is hard to say without seeing your code, but one possibility is you are using the serial port in such a way that it requires the Serial Monitor to be running. If you have any code that is writing out to the serial port, try commenting it out.

Dear M.,

I want to close the issue positive because the fault was mine. When the Arduino IDE runs open the serial communication and this communication remains open after the closing. But, if you unplug the usb of arduino micro an then plug it again , it works fine.

So, thank you very much. Your comment is very usefull.

closed 2023-02-13 by dimigav