MHeironimus/ArduinoJoystickLibrary #274: Problem with the USB cable disconneted

esportsim, 2023-04-12:
Hi, this is Carlo. Thanks for this amazing library.
I have a problem. I use the leonardo and when I unplug the usb cable from the PC the code cycle very slowly.
Do you know why? it's normal or it can be solved?
thanks you

MHeironimus, 2023-11-10:
If you are trying to communicate on the Serial port and you are not connected to the host computer (i.e., you unplug the USB cable), that can cause a slowdown in some cases.

esportsim, 2023-11-10:
There is a way to know if the usb cable is attached?

MHeironimus, 2023-11-10:
I have never done anything like that, but a quick search returned the following ideas:

- https://arduino.stackexchange.com/questions/438/how-does-this-code-line-detect-if-a-serial-interface-is-connected
- https://forum.arduino.cc/t/solved-detecting-serial-communication/500914/24
- https://forum.arduino.cc/t/checking-if-arduino-is-connected-serial-usb-to-a-pc/336007/26

Also, this may not be your issue. It was just a though.

esportsim, 2023-11-10:
Thank you very much.