MHeironimus/ArduinoJoystickLibrary #282: Simulate joystick unplug

valmontechno, 2024-01-02:
## Description of Issue
I'm looking for a way to completely stop the joystick emulation, like  it had been unplugged. I thought `Joystick.end()` was used for this but it doesn't seem to have any effect. The only way I found to achieve this is to reset the arduino. Is there any other less radical way?

## Technical Details

- Arduino Board: Arduino Micro
- Host OS: Window 10, Raspbian
- Arduino IDE Version: 2.2.1

## Sketch File that Reproduces Issue

~~~
#include <Joystick.h>

Joystick_ Joystick;

void setup() {
  delay(10000);
  pinMode(15, OUTPUT);
  digitalWrite(15, LOW);
}

void loop() {

}
~~~

## Wiring Details
Pins 15 and reset are connected.

nopbxlr, 2024-07-06:
Hello,
Bit of a late reply - but this closed issue has a USB reset function given by the repo author.
https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/288#issuecomment-2013648314
You should be able to simulate joystick unplug with this, with a bit of modification
Hope this will help you or anyone passing by

valmontechno, 2024-07-07:
Thanks, `UDCON` does indeed allow you to disable the USB port.

closed 2024-07-07 by valmontechno