MHeironimus/ArduinoJoystickLibrary #283: Samd21 boards

aelio, 2024-01-17:
Please indicate which board you would like to see supported.
Hi, I've made minor changes to get the library working on Trinket M0, probably it shall work on other samd boards as well. 
I used Gary Grewal's MIDIUSB as reference for accessing the platform's USB api.
It probably needs some graceful integration, here are the files
[dynamichid.zip](https://github.com/MHeironimus/ArduinoJoystickLibrary/files/13957338/dynamichid.zip)

relaxibus, 2024-01-17:
XIAO SAMD21? I didn't understood if the USB is a native USB or just TTL to USB.

aelio, 2024-01-17:
Native, via USB/PluggableUSB.h IC is ATSAMD21E18 32-bit Cortex M0+ on this board

relaxibus, 2024-01-17:
so using a SPI to ADC and I2C to GPIO expander we could have a nice solution for a joystick setup using the XIAO? 
Looking to substitute the https://www.leobodnar.com/shop/index.php?main_page=product_info&products_id=316

NNGB-kor, 2024-02-14:
Yes. This seems to work with Xioa's SAMD21.
I've compiled and uploaded a simple example.
It recognizes it as a joystick.

relaxibus, 2024-02-14:
Oh, that's really great!!! Could you send me the example? How did you uploaded to the Xiao?
May be you could also (I'm not a coder) try the above idea using native gpios or a SPI to ADC and I2C to GPIO expander?

carramar, 2024-10-12:
> Yes. This seems to work with Xioa's SAMD21. I've compiled and uploaded a simple example. It recognizes it as a joystick.

can you post how you got it to work please? im trying to get this to work on arduino zero (SAMD21) and cant get it to recognise as anything