MHeironimus/ArduinoJoystickLibrary #293: More than 1 analog joystick support

Industrial15, 2024-08-14:
So yeah, there is basically support for only 1 joystick and its working fine, but adding one more to the project breaks all away, i guess its will be usefull to people who want create something simmilar to playstation gamepad with their hands, i hope to get heared

MHeironimus, 2024-08-14:
Have you tried the MultipleJoystickTest example that is included in the library?

Industrial15, 2024-08-14:
> Have you tried the MultipleJoystickTest example that is included in the library?

Yes, but it doesn't work as expected, the code from the example uses different inputs 1 at a time such as throttle, brake, axle rotation, and so on. I would like to be able to have 2 inputs from two different joysticks that control their axes x/y, x1/y1, just the xbox/playstation gamepad for example

MHeironimus, 2024-08-14:
Sorry, I am not sure I understand what you are trying to do. This library does support creating multiple joysticks/game controllers, each with their own values. Reading values from input devices is not really "in scope" for this library. It is all about taking those input, once they have been read in, and sending them to the host computer in a format that it expects a joystick/game controller HID to be in.

If you post some sample code along with an explanation of how this code is not doing what you expect, someone might be able to help.

Industrial15, 2024-08-15:
i mean that i connect 2 joystick modules to my arduino board, and i want to emulate 1 device with 2 joysticks, like a regular gamepad, i hope i sayed it correctly