MHeironimus/ArduinoJoystickLibrary #295: INCOMPATIBLE WITH ARDUINO ZERO

carramar, 2024-10-12:
## Description of Issue
library claims to work with all native usb arduinos but is utterly incompatible with arduino zero which has native usb
## Technical Details

- Arduino Board (e.g. Arduino Leonardo): 
- Host OS (e.g. Windows 10):
- Arduino IDE Version (e.g. 1.8.3):

## Sketch File that Reproduces Issue

~~~
Sketch file goes here (if applicable)
~~~

## Wiring Details
Any pin wiring details that may be relevant.

## Additional context
Add any other context about the problem here.

per1234, 2024-10-12:
Duplicate of:

- https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/51
- https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/99
- https://github.com/MHeironimus/ArduinoJoystickLibrary/pull/101
- And your own https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/296

@carramar please close your duplicate issues to avoid cluttering up the issue tracker.

carramar, 2024-10-12:
I'll close the issue when it's no longer an issue, if there are three other
issues open about the same thing (and given that this issue has been known
for over 5 years) then it might be worth fixing

MHeironimus, 2024-10-14:
Actually the library only claims to work with the boards listed on the following wiki page: https://github.com/MHeironimus/ArduinoJoystickLibrary/wiki/Supported-Boards

The author (me) makes no claims that this library will work for anyone other than himself. :) I have posted this code as a community service and update is as I am able.

Feel free to dive into the code to see if you can get it to work with an Arduino Zero. There is a PR out there you can try https://github.com/MHeironimus/ArduinoJoystickLibrary/pull/101 that claims to support Arduino Zero. I have not had a chance to validate it, but it might work for your needs.

I'll leave this open so that once you have tried out PR 101, you can post whether or not it worked for you.

MHeironimus, 2024-10-14:
Here is another PR you can try out: https://github.com/MHeironimus/ArduinoJoystickLibrary/pull/262 that also is for supporting the Arduino Zero.

closed 2025-01-18 by MHeironimus