MHeironimus/ArduinoJoystickLibrary #297: Button Press Not Detected in Game Controller

msturge, 2024-10-29:
## Description of Issue
I have written a sketch using this library along with an Arduino Micro to crate a button box with 7 buttons in a 2 column 5 row configuration. When listening to the serial output I can detect the button presses correctly and then set Joystick.setButton(<button_character>, 1);

I see the correct output in the Serial Watcher, but when checking JoyToKey or the native windows Game Controllers window the button presses are not detected. 

The numbering in the keys array is a little messey due to the way I physically wired the buttons. 

## Technical Details

- Arduino Micro
- Windows 11
- Arduino IDE Version  2.3.3
- Library Version 2.1.1

## Sketch File that Reproduces Issue
```
#include <Keypad.h>
#include <Joystick.h>

#define NUM_BUTTONS 10

const byte ROWS = 5; // Four rows
const byte COLS = 2; // Four columns

char keys[ROWS][COLS] = {
  {'2', '7'},
  {'3', '8'},
  {'4', '9'},
  {'0', '5'},
  {'1', '6'}
};

byte rowPins[ROWS] = {5, 6, 7, 8, 9}; // Connect to the row pinouts of the keypad
byte colPins[COLS] = {10, 11}; // Connect to the column pinouts of the keypad

Keypad buttbx = Keypad( makeKeymap(keys), rowPins, colPins, ROWS, COLS);

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_JOYSTICK,
                   NUM_BUTTONS, 0,                  // Button Count, Hat Switch Count
                   false, false, false,     // X and Y, but no Z Axis
                   false, false, false,   // No Rx, Ry, or Rz
                   false, false,          // No rudder or throttle
                   false, false, false);  // No accelerator, brake, or steering

void setup() {
  Serial.begin(9600);
  Joystick.begin();
  Serial.println("Serial Working");
}

void loop() {
  CheckAllButtons();
}

void CheckAllButtons(void) {
  if (buttbx.getKeys()){
    for (int i = 0; i < LIST_MAX; i++){
      if ( buttbx.key[i].stateChanged ){
        switch (buttbx.key[i].kstate) {
          case PRESSED:
          case HOLD:
            Serial.println(buttbx.key[i].kchar);
            Joystick.setButton(buttbx.key[i].kchar, 1);
            break;
          case RELEASED:
          case IDLE:
            Joystick.setButton(buttbx.key[i].kchar, 0);
            break;
        }
      }
    }
  }
}
```
~~~
Sketch file goes here (if applicable)
~~~

## Wiring Details
Cols are wired to pins 10 and 11
Rows are wired to pins 5,6,7,8,9

MHeironimus, 2024-10-29:
I suspect buttbx.key[i].kchar is not returning a number between 0 and 9. My guess is it is returning a char '0' (which would be 48), '1' (which is actually 49), etc .

msturge, 2024-10-29:
That's a very good point.  I'll ensure the variables are int typed. Thank you

msturge, 2024-11-03:
Just checked the code, that was indeed the issue. Thank you for your help!

closed 2024-11-03 by msturge