MHeironimus/ArduinoJoystickLibrary #305: Libraries don't function :-(

foxhunter-73, 2025-01-24:
I'm working on windows 10, and i use the ide 2.x and also test on 1.8.19
I easly acquire the library and try to compile an example ,
dont' function !!
The error is:

`
In file included from D:\nas\cloud\cloud\arduino_sck\libraries\Joystick\src/Joystick.h:24:0,
                 from C:\Users\riccardo\AppData\Local\Temp\arduino_modified_sketch_477170\sketch_jan24a.ino:1:
D:\nas\cloud\cloud\arduino_sck\libraries\Joystick\src/DynamicHID/DynamicHID.h:107:1: error: expected class-name before '{' token
 {
 ^
exit status 1
Errore durante la compilazione per la scheda Arduino Leonardo.

`

Is the same always on the different ide

foxhunter-73, 2025-01-26:
If i compile it like Arduino Zero , the library Was good ( i haven't tested it ), but if i compile like Leonardo the error about Up happened.
I'm not an expert but i understand that it happened when
#include "DynamicHID/DynamicHID.h"
are active, if i comment the error don't happened

Lunahri17, 2025-02-21:
check this issue [https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/38](url)

You need to copy the dinamychid folder to the libraries and import on your main .ino file

foxhunter-73, 2025-02-21:
Thanks, but don't function!I was just try....😭😭 However i use nicohood library that function well !💪