MHeironimus/ArduinoJoystickLibrary #306: Missing from the Arduino Library Registry

dmadison, 2025-02-22:
Please add this library to the official [Arduino Library Registry](https://github.com/arduino/library-registry) so that it can be installed through the IDE and linked as a dependency in other libraries.

To add this library to the registry, all you would need to do is:
* Fork the [Arduino Library Registry repository](https://github.com/arduino/library-registry)
* Add the URL of this library to [the repositories.txt file](https://github.com/arduino/library-registry/blob/main/repositories.txt)
* Submit a [pull request](https://github.com/arduino/library-registry/compare) to the registry with your change

MHeironimus, 2025-02-22:
I hope to do this eventually, but the current name of the library conflicts with another library that is already in the registry. I will need to rename it first, which will be a breaking change for everyone who currently uses it.

dmadison, 2025-02-22:
I think the only part that needs to be renamed is the `name` field of the `library.properties` file. The rest of the library, including the header file, can stay the same. That should only affect the name of the library in the IDE (e.g. in the libraries manager and the 'examples' menu). All existing code would still be compatible.

Changing the name field to something like "Joystick Library", albeit redundant, would fix the conflict without any side effects.

IvoryRubble, 2025-05-02:
Any update? I (as user) would appreciate this awesome lib to be in Arduino Library Registry