MHeironimus/ArduinoJoystickLibrary #307: Buttons do not work on 2nd ps2 controller

takeriveter28, 2025-03-18:
As the title says,

I've connected two dualshock 2 controllers to a pro micro.

The 1st controller works flawlessly, including analog sticks.

The 2nd controller's buttons do not work, only the analog sticks work.

Here's my code:

```
#include <PsxControllerBitBang.h>
#include <Joystick.h>

// Pin assignments for the first controller
const byte PIN_PS2_ATT1 = 6;
const byte PIN_PS2_CMD1 = 7;
const byte PIN_PS2_DAT1 = 8;
const byte PIN_PS2_CLK1 = 5;

// Pin assignments for the second controller
const byte PIN_PS2_ATT2 = 15;
const byte PIN_PS2_CMD2 = 14;
const byte PIN_PS2_DAT2 = 16;
const byte PIN_PS2_CLK2 = A0;

const unsigned long POLLING_INTERVAL = 1000U / 50U;
#define ENABLE_SERIAL_DEBUG

#define JOYSTICK_COUNT 2  // We are working with two PS2 controllers

// Declare Joystick array for two controllers
Joystick_ Joystick[JOYSTICK_COUNT] = {
  Joystick_(0x03, JOYSTICK_TYPE_JOYSTICK, 16, 0, true, true, false, true, true, false, false, false, false, false, false),  // First PS2 controller
  Joystick_(0x04, JOYSTICK_TYPE_JOYSTICK, 16, 0, true, true, false, true, true, false, false, false, false, false, false)   // Second PS2 controller
};

PsxControllerBitBang<PIN_PS2_ATT1, PIN_PS2_CMD1, PIN_PS2_DAT1, PIN_PS2_CLK1> psx1;
PsxControllerBitBang<PIN_PS2_ATT2, PIN_PS2_CMD2, PIN_PS2_DAT2, PIN_PS2_CLK2> psx2;

#ifdef ENABLE_SERIAL_DEBUG
  #define dstart(spd) do {Serial.begin (spd); while (!Serial) {digitalWrite (LED_BUILTIN, (millis () / 500) % 2);}} while (0);
  #define debug(...) Serial.print (__VA_ARGS__)
  #define debugln(...) Serial.println (__VA_ARGS__)
#else
  #define dstart(...)
  #define debug(...)
  #define debugln(...)
#endif

boolean haveController1 = false;
boolean haveController2 = false;

void setup() {
  // Initialize each joystick
  for (int i = 0; i < JOYSTICK_COUNT; i++) {
    Joystick[i].begin(false);  // Set to false for manual mode
    Joystick[i].setXAxisRange(ANALOG_MIN_VALUE, ANALOG_MAX_VALUE);
    Joystick[i].setYAxisRange(ANALOG_MIN_VALUE, ANALOG_MAX_VALUE);
    Joystick[i].setRxAxisRange(ANALOG_MIN_VALUE, ANALOG_MAX_VALUE);
    Joystick[i].setRyAxisRange(ANALOG_MIN_VALUE, ANALOG_MAX_VALUE);
  }

  pinMode(LED_BUILTIN, OUTPUT);
  dstart(115200);  // Start serial communication for debugging
  debugln(F("Ready!"));
}

void loop() {
  static unsigned long last = 0;

  if (millis() - last >= POLLING_INTERVAL) {
    last = millis();

    // Handle first controller (psx)
    if (!haveController1) {
      if (psx1.begin()) {
        debugln(F("Controller 1 found!"));
        if (!psx1.enterConfigMode()) {
          debugln(F("Cannot enter config mode"));
        } else {
          if (!psx1.enableAnalogSticks()) {
            debugln(F("Cannot enable analog sticks"));
          }
          if (!psx1.exitConfigMode()) {
            debugln(F("Cannot exit config mode"));
          }
        }
        haveController1 = true;
      }
    } else {
      if (!psx1.read()) {
        debugln(F("Controller 1 lost"));
        haveController1 = false;
      } else {
        byte x, y;
        // Update button states for controller 1
        Joystick[0].setButton(0, psx1.buttonPressed(PSB_TRIANGLE));
        Joystick[0].setButton(1, psx1.buttonPressed(PSB_CROSS));
        Joystick[0].setButton(2, psx1.buttonPressed(PSB_SQUARE));
        Joystick[0].setButton(3, psx1.buttonPressed(PSB_CIRCLE));
        Joystick[0].setButton(4, psx1.buttonPressed(PSB_L1));
        Joystick[0].setButton(5, psx1.buttonPressed(PSB_R1));
        Joystick[0].setButton(6, psx1.buttonPressed(PSB_L2));
        Joystick[0].setButton(7, psx1.buttonPressed(PSB_R2));
        Joystick[0].setButton(8, psx1.buttonPressed(PSB_SELECT));
        Joystick[0].setButton(9, psx1.buttonPressed(PSB_START));
        Joystick[0].setButton(10, psx1.buttonPressed(PSB_L3));
        Joystick[0].setButton(11, psx1.buttonPressed(PSB_R3));
        Joystick[0].setButton(12, psx1.buttonPressed(PSB_PAD_UP));
        Joystick[0].setButton(13, psx1.buttonPressed(PSB_PAD_DOWN));
        Joystick[0].setButton(14, psx1.buttonPressed(PSB_PAD_LEFT));
        Joystick[0].setButton(15, psx1.buttonPressed(PSB_PAD_RIGHT));

        // Update axes for controller 1
        if (psx1.getLeftAnalog(x, y)) {
          Joystick[0].setXAxis(x);
          Joystick[0].setYAxis(y);
        }
        if (psx1.getRightAnalog(x, y)) {
          Joystick[0].setRxAxis(x);
          Joystick[0].setRyAxis(y);
        }

        // Send the state for controller 1
        Joystick[0].sendState();
      }
    }

    // Handle second controller (psx2)
    if (!haveController2) {
      if (psx2.begin()) {
        debugln(F("Controller 2 found!"));
        if (!psx2.enterConfigMode()) {
          debugln(F("Cannot enter config mode"));
        } else {
          if (!psx2.enableAnalogSticks()) {
            debugln(F("Cannot enable analog sticks"));
          }
          if (!psx2.exitConfigMode()) {
            debugln(F("Cannot exit config mode"));
          }
        }
        haveController2 = true;
      }
    } else {
      if (!psx2.read()) {
        debugln(F("Controller 2 lost"));
        haveController2 = false;
      } else {
        byte x, y;
        // Update button states for controller 2
        Joystick[1].setButton(0, psx2.buttonPressed(PSB_TRIANGLE));
        Joystick[1].setButton(1, psx2.buttonPressed(PSB_CROSS));
        Joystick[1].setButton(2, psx2.buttonPressed(PSB_SQUARE));
        Joystick[1].setButton(3, psx2.buttonPressed(PSB_CIRCLE));
        Joystick[1].setButton(4, psx2.buttonPressed(PSB_L1));
        Joystick[1].setButton(5, psx2.buttonPressed(PSB_R1));
        Joystick[1].setButton(6, psx2.buttonPressed(PSB_L2));
        Joystick[1].setButton(7, psx2.buttonPressed(PSB_R2));
        Joystick[1].setButton(8, psx2.buttonPressed(PSB_SELECT));
        Joystick[1].setButton(9, psx2.buttonPressed(PSB_START));
        Joystick[1].setButton(10, psx2.buttonPressed(PSB_L3));
        Joystick[1].setButton(11, psx2.buttonPressed(PSB_R3));
        Joystick[1].setButton(12, psx2.buttonPressed(PSB_PAD_UP));
        Joystick[1].setButton(13, psx2.buttonPressed(PSB_PAD_DOWN));
        Joystick[1].setButton(14, psx2.buttonPressed(PSB_PAD_LEFT));
        Joystick[1].setButton(15, psx2.buttonPressed(PSB_PAD_RIGHT));

        // Update axes for controller 2
        if (psx2.getLeftAnalog(x, y)) {
          Joystick[1].setXAxis(x);
          Joystick[1].setYAxis(y);
        }
        if (psx2.getRightAnalog(x, y)) {
          Joystick[1].setRxAxis(x);
          Joystick[1].setRyAxis(y);
        }

        // Send the state for controller 2
        Joystick[1].sendState();
      }
    }
  }
}
```

I'm using Arduino IDE v2.3.4.
I'm using the latest git version of all 3 libraries: PsxNewLib, DigitalIO, ArduinoJoystick

I got my first pro micro and arduino UNO yesterday, so I'm an absolute newb at this.
Any help or suggestions regarding this is much appreciated :D

MHeironimus, 2025-03-18:
To debug this, I would start by writing out the `psx2.buttonPressed` values read in to the serial debugger to see if they are recording the button presses accurately.

takeriveter28, 2025-03-19:
As per your advice, I added this code snippet:

```
debug(F("Square button: "));
debugln(psx2.buttonPressed(PSB_SQUARE), BIN);
debug(F("L1 button: "));
debugln(psx2.buttonPressed(PSB_L1), BIN);
debug(F("L2 button: "));
debugln(psx2.buttonPressed(PSB_L2), BIN);
debug(F("SELECT button: "));
debugln(psx2.buttonPressed(PSB_SELECT), BIN);

// Send the state for controller 2
Joystick[1].sendState();
}
```

Now, pressing all of these 4 buttons at once.
Output:

```
Square button: 1
L1 button: 1
L2 button: 1
SELECT button: 1
```

So the raw data is being received by the pro micro.
I even tested the controller separately and even then the issue only occurs in the "2nd slot" / psx2.

Is there some "seperation" needed that will make the 2 controllers can show up and separately in the pc, or is that handled by default ?

MHeironimus, 2025-03-22:
> I'm using Debian Bookworm as my OS, I don't know if that play's a role in this or not.

Yes, that is an important detail. Linux does not (by default) support multiple joysticks on a single USB device. It can be coaxed into allowing it (sometimes). See the Linux Support section of the wiki (https://github.com/MHeironimus/ArduinoJoystickLibrary/wiki/FAQ#linux-support) for more details.

takeriveter28, 2025-03-22:
Oh, my bad. I just assumed that it worked the same way on both windows and linux.

I'll check it out and report my findings.

takeriveter28, 2025-03-22:
Update:

The issue has been solved !
It was the usb quirks feature.
I enabled it and now I can connect even 5 controllers simultaneously (though the inputs get a little choppy, probably because I used the same 3 signal lanes for all 5 controllers).

Thanks for the help ! :D

MHeironimus, 2025-03-22:
Glad to hear you were able to get it working.

closed 2025-03-22 by MHeironimus